udacity/self-driving-car-sim #2: Unity just freezes during import

idoan, 2017-02-08:
1) while project is getting imported progress bar freezes on " importing small assets" step. 

This is how I load the project:

- clone the repo with git lfs followingly:

`git lfs clone https://github.com/udacity/self-driving-car-sim.git`

Load the project from "Open" tab on the first screen. ( Unlike what you wrote on the Readme, there is no " Load up the cloned folder self-driving-car-sim inside Unity as a new project " option, either create New from blank or Open the existing one)

then it starts loading and then it stuck at "importing small assets" step on progress bar. 

I am on mac.
intel hd 530
16 gb ram

can you guys help me ?

idoan, 2017-02-09:
having killing the app when it stuck on "importing small assets", I kept loading the same project and it still got stuck a few more times, yet at last it loaded. It seems to be working fine now. however, there might be further issues on this, I will post an update here after I test this process sufficient times. 

Thanks

snuffysasa, 2017-02-10:
Yea Unity crashes for me when trying to import assets.
I am on Mac OS 10.12

Crashed Thread:        0  CrBrowserMain  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGABRT)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000060
Exception Note:        EXC_CORPSE_NOTIFY

snuffysasa, 2017-02-10:
Crashes for me when importing Assets on Ubuntu as well.

Tbyo, 2017-02-13:
Crashes for me when importing small assets on Ubantu14.04 as well. 
The fatal error is : Out of memory! UnityEditior: AssetPreviewUpdater:CreatePreviewForAsset(Object,Object[],String).
However, plenty of my system memory space is still left. In addition, I can create a new project using Unity.

domluna, 2017-02-13:
What version of Unity are you folks using? I'm using 5.5.1 on Ubuntu 16.04 and it works well.

snuffysasa, 2017-02-13:
I am also on 5.5.1,   Ubuntu 16.04

snuffysasa, 2017-02-13:
Unity works fine, but when importing this project in particular it crashes.

domluna, 2017-02-13:
Hmmm, actually I'm using elementaryos 0.4 which is based off ubuntu 16.04. But I don't think that would make a difference.

snuffysasa, 2017-02-13:
Yea seems unlikely.... Do you know how I can see my crash details?

domluna, 2017-02-13:
Not sure. https://forum.unity3d.com/threads/unity-on-linux-release-notes-and-known-issues.350256/#post-2772944 the one I downloaded was from here. Search for 5.5.1.

snuffysasa, 2017-02-13:
I have the same version, reinstalled from the link you gave me.   It only crashes when loading certain assets from the the project. Crashes on my macbook, and on ubuntu desktop.  When importing the plugins folder, or importing the prefabs folder within 1_Selfdrivingcar

Tbyo, 2017-02-14:
Unity works fine for me when I upgrade the unity version from 5.1.0 to 5.5.1.  Thanks @domluna

abmaurya, 2017-02-16:
Crashed on Unity 5.3.4 (on Windows), but works fine after 5.5.1 Update. But there is another problem with parsing PNG files in the sim folder  (these are actually text files but have been given the extension **PNG**, don't know why).

purudpd, 2017-12-03:
Were you able to fix the import issue with PNG files ?

mvirgo, 2019-01-11:
No recent comments (looks like main issue was resolved), so closing out.

closed 2019-01-11 by mvirgo