udacity/self-driving-car-sim #3: Missing assembly?

stevenhedges, 2017-02-09:
I've imported the project into Unity, but when trying to view a scene I get the following error:

Assets/1_SelfDrivingCar/Scripts/CommandServer.cs(4,7): error CS0246: The type or namespace name `SocketIO' could not be found. Are you missing an assembly reference?

tawnkramer, 2017-02-09:
Check your Assets/SocketIO/Scripts/SocketIO and make sure there are files there. Sounds like they might have gone missing for you? Also which scene? The one's I tried seem to work.

vlchz, 2017-06-09:
So how do you add the files? Under "Scripts" there are no SocketIO files

vlchz, 2017-06-09:
I fixed it now it's giving me 16 errors saying TerrainData is missing splat texture 0....The number at the end is different numbers

clemens-tolboom, 2017-08-07:
> Assets/1_SelfDrivingCar/Scripts/CommandServer.cs(4,7): error CS0246: The type or namespace name `SocketIO' could not be found. Are you missing an assembly reference?

@vlchz how did you fixed this missing SocketIO?

clemens-tolboom, 2017-08-07:
My temp solution is to download the ZIP file which is a PITB. Is the git checkout command of git client causing this?

clemens-tolboom, 2017-08-07:
I'm on a Mac 10.11.6

clemens-tolboom, 2017-08-07:
I get

> Resolving deltas: 100% (1987/1987), done.
Downloading Assets/1_SelfDrivingCar/Sprites/Background.png (7.2 MB)
Error downloading object: Assets/1_SelfDrivingCar/Sprites/Background.png (36e4565): Smudge error: Error downloading Assets/1_SelfDrivingCar/Sprites/Background.png (36e4565319a778a53538a2c1419174cac59597859e5a8616c1b94a08d6221ac3): batch response: This repository is over its data quota. Purchase more data packs to restore access.
>
>Errors logged to /Users/clemens/Documents/projects/unity/X/.git/lfs/objects/logs/20170807T160036.124375362.log
Use `git lfs logs last` to view the log.

So the trouble is with Quota as described in https://help.github.com/articles/about-storage-and-bandwidth-usage/

I've tried to run the https://github.com/udacity/self-driving-car-sim/archive/master.zip which seemed to fail too for other reason so I downloaded the Mac executable (beta-simulator-mac.zip)

I guess we have found the problem. To many git clones per month.

mvirgo, 2019-01-11:
No recent comments, and looks like it was mainly due to too many git clones; closing out.

closed 2019-01-11 by mvirgo