udacity/self-driving-car-sim #6: Generated CSV file is locale-specific

kenshin23, 2017-02-11:
While generating data for the Behavioral Cloning project, I found something interesting (and highly annoying) with the beta simulator in Linux (Ubuntu 16.10 x64). I was trying to generate new data and when finished recording, I tried to load my CSV file only to find out that the values didn't make sense. When checking it, I found that all the values had a comma instead of a period as the decimals separator (I live in Venezuela so my PC's locale is set as es_VE), so I got unintended extra columns in the file.
Running the simulator as $ `LC_ALL ./beta_simulator.x86_64 &` serves as a workaround.

tomi92, 2017-04-01:
I also have this problem.

AdrianP-, 2017-09-30:
Same here. Fixed with @tdanyluk command