udacity/self-driving-car-sim #10: Can I try this projects on my Ubuntu?

yi2ha, 2017-02-15:
Can I try this project on my Ubuntu?But Unity  just on win or mac.
And I try download beta-simulator-linux.zip ,then get beta_simulator.x86_64,beta_simulator.x86,beta_simulator_Data.

But I can't run it. Is that some steps or files I miss ? I love this project, hope for your help!

ilyes495, 2017-02-22:
Make sure the binary file has "execute permission"...
The easy way , right click in the file, then properties, go to the permissions tab and click in the "Allow executing file as program".
[https://i.stack.imgur.com/sFi9x.png](url)

weiy1991, 2017-03-13:
Hi,  

I try to use this simulator on my ubuntu, I could run the linux_sim.84_64 and configure the param , but after I press the 'OK' button, the configuration closed and no software launched again, there's no error information. I don't know how to solve the problem.

ARastogi19, 2017-04-17:
I have the same exact issue.

kalyanperi, 2017-07-03:
Hey guys, Im in same position. Did u guys find any solution?

MichaelYang21, 2017-09-23:
I have the same issue. Any solution for this one? Thanks

robogeekcanada, 2017-10-10:
Any help will be appreciated. Thank you in advance.

sgzzgit, 2017-10-19:
I have the same issue, feel sad.

robogeekcanada, 2017-10-19:
I spent a lot of time on this and it's really tricky to make it work in Ubuntu. I am sure it's possible but I decided to start fresh with a Windows Server and it's working flawlessly. I also tried in AWS and Google Cloud with windows and works well.

bhaveshkumarraj, 2017-12-30:
For those who are still facing problem in running this simulator on Ubuntu.
Use the older stable version of simulator(link given below). I tried on Ubuntu 16.04 and it worked.
https://d17h27t6h515a5.cloudfront.net/topher/2016/November/5831f0f7_simulator-linux/simulator-linux.zip

laygond, 2018-01-10:
Hello bhaveshkumarraj are you able to run Term 2 - Project 5 - MPC with this simulator version? I get the following error and I want to know if the error is related to the simulator or something else.

laygond@laygond-VPCS111FM:~/karolmajekMPC/build$ ./mpc
factor_cte    2.000000
factor_epsi   5.000000
factor_v      0.100000
factor_steering 1000.000000
factor_throttle 0.100000
factor_seq_steering 30000.000000
factor_seq_throttle 1.000000
Listening to port 4567
Connected!!!
terminate called after throwing an instance of 'std::domain_error'
  what():  type must be array, but is null
Aborted (core dumped)

bhaveshkumarraj, 2018-01-10:
@laygond My requirement was to collect driving data to train the model. I just downloaded pre-compiled simulator using the link mentioned below. 
https://d17h27t6h515a5.cloudfront.net/topher/2016/November/5831f0f7_simulator-linux/simulator-linux.zip
I haven't worked on CarND-MPC-Project so I can't say anything about it.

baumanab, 2018-02-19:
Closed since we have moved far beyond beta.

closed 2018-02-19 by baumanab