udacity/self-driving-car-sim #16: Camera distortion correction needed?

AhmedAnwarHafez, 2017-02-26:
I'm interested in applying all the knowledge I gained in the Advanced Lane Detection project and put it into the Car Simulator.

How do I know that the left, right and center camera are corrected or non-corrected?

AhmedAnwarHafez, 2017-03-12:
I managed to find the camera matrix and distortion coefficients by downloading the source code project and unity and then took many screenshots of the chessboard.

Here is a notebook that I wrote on how I did it.
https://github.com/AhmedAnwarHafez/SDC-Advanced-Lane-Finding-In-Simulator/blob/master/Camera%20Caliberation%20for%20Udacity's%20Car%20Simulator.ipynb

baumanab, 2018-02-19:
That is awesome, way to take the initiative!

closed 2018-02-19 by baumanab