udacity/self-driving-car-sim #18: Simulator crashes on startup inside a virtual machine

mukilk7, 2017-02-28:
I'm trying to run the linux build of the simulator in an Amazon EC2 VM. I get the initial screen that asks for screen resolution and graphics quality, but not matter what options I choose for those two, my application ends crashing and core dumping. I ran strace and it showed me that the program received SIGABRT. I also tried the following things:

* I installed gnome desktop manager and vnc to ensure that I can interact with GUI applications. 
* I did x-forwarding over SSH from the Amazon EC2 VM to a local linux VM on my laptop.

None of these options helped with the crash. How do I get to the bottom of this? I'm beginning to sense that this might be a problem with the unity3d engine.

coventry, 2017-03-30:
I hit this bug, too.

Steps to reproduce:

1. Start an AWS t2.medium instance with 20G of memory, running ubuntu 16.04.
2. Run these commands to install unity:
```bash
wget http://beta.unity3d.com/download/e348e673a4c6/unity-editor_amd64-5.6.0xf1Linux.deb
sudo dpkg -i unity-editor_amd64-5.6.0xf1Linux.deb  # Fails, but lets apt know what it needs.
sudo cp unity-editor_amd64-5.6.0xf1Linux.deb /var/cache/apt/archives/
sudo apt-get update
sudo apt-get install -f
sudo apt-get install unity-editor vnc4server unzip sawfish-lisp-source
```
3. Start vnc with `vncserver -geometry 1600x1200`.  Forward the vnc port by running `ssh -L 5901:localhost:5901 d2` on my local machine.  Connect to the vnc session with remmina. Run `sawfish &` for a basic window manager.
3. Run these commands to install the simulator:
```bash
wget https://d17h27t6h515a5.cloudfront.net/topher/2017/February/58983558_beta-simulator-linux/beta-simulator-linux.zip
unzip beta-simulator-linux.zip
```
4. Start unity with `/opt/Unity/Editor/Unity`, log in, click "Open" in the top right-hand corner, choose `beta_simulator_linux`.

The shell reports `Aborted (core dumped)`.  With `ulimit -c unlimited`, this drops a core in `beta_simulator_linux`, but the binary has no symbols so it's pretty uninformative.

Could this be happening because there's no video card?  Is there any way to check, short of building unity from source?

coventry, 2017-03-30:
I believe this is due to an incompatibility between Unity and VNC.  I got Unity working on my local workstation, then tried to run it in a local VNC session, and it dumped core in the same way.  I'm going to open a new issue.

jgilbert42, 2017-06-04:
I'm having the same issue with Ubuntu 16.04.2 in Parallels on a Mac.

ModiGit, 2017-11-15:
Hello, 

I had the same issue with ubuntu 14.04 and unity-editor 5.5.1. I updated to unity 5.6.0 and that fixed the problem.
Here is what I did:
1- sudo apt-get install gdebi
2-  wget http://beta.unity3d.com/download/e348e673a4c6/unity-editor_amd64-5.6.0xf1Linux.deb
3-  sudo gdebi unity-editor_amd64-5.6.0xf1Linux.deb 

After that it unity worked perfectly. Hope it helps someone.

Thanks...

yiqin, 2017-11-20:
I'm also have the same issue in VMWare on a Mac.