udacity/self-driving-car-sim #20: The sim can't run on my ubuntu

weiy1991, 2017-03-11:
Hi,

I try to run the sim on my ubuntu, and I have changed the permission to allow the sim to be running as a program, but the sim can only start with the configuration step. After I confirm the configuration of the sim, the sim just close. 

can anybody know this?

mwolfram, 2017-05-12:
Same here, I'm running **Ubuntu 14.04 LTS**. 

How to reproduce:
1. Download the tgz and unpack it
2. Make sure binaries are executable
3. Execute term2_sim.x86_64
4. Check the config (seemed fine on first run, but I also tried changing it), press OK
5. Program exits without error messages, and with the following output:

`mwolfram@ubuntu:~/udacity/sdcnd/term2/term2_sim_linux$ ./term2_sim.x86_64`
`Set current directory to /home/mwolfram/udacity/sdcnd/term2/term2_sim_linux`
`Found path: /home/mwolfram/udacity/sdcnd/term2/term2_sim_linux/term2_sim.x86_64`
`Mono path[0] = '/home/mwolfram/udacity/sdcnd/term2/term2_sim_linux/term2_sim_Data/Managed'`
`Mono path[1] = '/home/mwolfram/udacity/sdcnd/term2/term2_sim_linux/term2_sim_Data/Mono'`
`Mono config path = '/home/mwolfram/udacity/sdcnd/term2/term2_sim_linux/term2_sim_Data/Mono/etc'`
`displaymanager : xrandr version warning. 1.5`
`client has 8 screens`
`displaymanager screen (0)(Virtual1): 1920 x 1080`
`Using libudev for joystick management`
`Importing game controller configs`

This is the same behaviour as with the simulator in term 1 (behavioural cloning project), maybe it's related.

rohts-patil, 2017-06-01:
How did you make sure that binaries are executable?

mwolfram, 2017-06-02:
Hi Rohit,

I ran the simulator on Windows and connected to my VM, which is running
Ubuntu. However, if you want to render a binary executable on Ubuntu, just
type:

sudo chmod +x <my_binary>

kind regards

ricgu8086, 2017-06-19:
I have the same problem. It can be reproduced exactly as @mwolfram commented.

lelordoftech, 2017-07-12:
Hi all,

I found the problem is OpenGL version in Ubuntu 14.04 VMware by default is 2.1.
But the simulator request OpenGL version higher.
In my case:
- VMware Player 7.0
- Ubuntu 14.04 LTS

I fixed by:
- Step 1: upgrade VMware Player to VMware Workstation 12 Player
- Step 2: disable "Accelerate 3D graphics” in VMware Player
- Step 3: upgrade vmwgfx kernel module of Ubuntu 14.04
Download:
# git clone https://anongit.freedesktop.org/git/mesa/vmwgfx.git
Build:
# cd vmwgfx
# make
Replace original vmwgfx.ko
# sudo mv /lib/modules/3.19.0-80-generic/kernel/drivers/gpu/drm/vmwgfx/vmwgfx.ko /lib/modules/3.19.0-80-generic/kernel/drivers/gpu/drm/vmwgfx/vmwgfx_old.ko
# sudo cp vmwgfx.ko /lib/modules/3.19.0-80-generic/kernel/drivers/gpu/drm/vmwgfx/vmwgfx.ko
# sudo depmod -a
- Step 4: check OpenGL version
# glxinfo | grep "OpenGL version"
The result will be:
**OpenGL version string: 3.0 Mesa 10.3.2**
And now I can use the simulator perfectly.

Reference:
- https://www.mesa3d.org/vmware-guest.html
- https://bbs.archlinux.org/viewtopic.php?id=202713

spencerh-b, 2017-08-29:
This solution did not seem to work for me. I'm running Ubuntu 14.04 without any VM. My only issue seems to be related to compiler errors with in unity. Many of them are related to "Failed to load package manager" and "Are you missing an assembly reference?" Is this an Ubuntu issue or sim Issue?

lelordoftech, 2017-08-29:
Hi @spencerh-b ,
I think your issue is different with my issue.
I used Ubuntu 14.04 without any VM look like you and it works perfectly.
My issue just happened when I use Ubuntu in VM.

Btw I have 1 suggestion, you should upgrade to the latest Ubuntu release, maybe 17.04.
Because when you move onto Term 2, Term 3, you have to upgrade many thing, install many lib, and Ubuntu 17.04 has everything for us :)
Trust me, I took a lot of time to setup environment in Ubuntu 14.04 (without any VM) for Term 2. After 1, 2 days, I decided to upgrade to Ubuntu 17.04 in order to resolve many issues in Ubuntu 14.04.
Because I'm not a expert in Ubuntu and I don't want to take too much time for setting up environment :)

asimay, 2017-09-18:
very good. i solve my problem. Thanks.
here: the modules/3.19.0-80-generic/ directory maybe different from yours. need to change.

sudo mv /lib/modules/3.19.0-80-generic/kernel/drivers/gpu/drm/vmwgfx/vmwgfx.ko /lib/modules/3.19.0-80-generic/kernel/drivers/gpu/drm/vmwgfx/vmwgfx_old.ko

robogeekcanada, 2017-10-10:
I think I'm getting closer but after implementing the suggested fix, I get this error when i run:
glxinfo | grep render

Error: couldn't find RGB GLX visual or fbconfig

npv0, 2017-12-31:
This was an issue for my friend, it was solved by following the step 1 and 2 mentioned by  @lelordoftech .Step 3 was not followed.
VM : VM ware workstation player 12
ISO: ubuntu-16.04.3-desktop-amd64
Network : Bridged
Display : Turn off "accelerate 3d graphics"

asimay, 2018-01-01:
or else, I published a paper before: 
https://medium.com/@asimay_y/how-to-solve-the-testing-problems-of-udacity-simulator-between-windows-and-vmware-linux-d7cc620ef1a2

you can access this.

tsugiar, 2018-10-03:
I have similar problem on my Ubuntu 14.04 that runs on VMWare in Win7, where simulator disappears without error message. I try to follow suggestion from this thread about updating vmgfx, but I stuck with compilation error. 

Then, I try alternate approach where I put simulator in windows environment and project code in Ubuntu side. I tried to follow "asimay", however, I can't find "Virtual Network Editor" on my VMWare player to do port-forwarding. Instead of using "Virtual Network Editor" from VMWare, I use windows port forwarding command "netsh".

Below is the procedure that I follow to use netsh command:
a.  I set my Ubuntu configured as NAT and type ifconfig to get its IP address
    (e.g. in my case I get 192.168.203.128 for Ubuntu IP address)
b. Go to windows side, and open up dos-command prompt as "Admin-User"
c. Then, inside dos prompt type:
    netsh interface portproxy add v4tov4 listenport=4567 listenaddress=127.0.0.1 connectport=4567 connectaddress=192.168.203.128
(Replace address value of "connectaddress" with Ubuntu IP address that you get from ifconfig in step a)

By doing steps above, I am able to run simulator in windows side and interact with my project-code that resides in my Ubuntu side.

Note:  The port-forwarding set by netsh command in step c will persist in windows setup. So, I finish with
           all the work, I delete forwarding rule by issuing the command below:
   netsh interface portproxy delete v4tov4 listenport=4567 listenaddress=127.0.0.1

Hope this help,

jiehou, 2019-02-06:
hello,

i want to ask, does the latest simulator really work on Ubuntu 17 without a virtual machine? thanks