udacity/self-driving-car-sim #26: Term 2 Simulator Doesn't Run on Mac

vernor1, 2017-05-05:
I downloaded and unpacked term2_sim_mac.zip (release v1.0, May 3), but the app doesn't run, throwing the error: The application "term2_sim" can't be opened.

awbrown90, 2017-05-06:
Hi vernor1, tried the Term 2 simulator out on two different Macs without problems. Maybe the iOS version you are running is out of date?

vernor1, 2017-05-06:
Hi Aaron,
I'm running the latest macOS 10.12.4 on MacBook Pro 2014, and all other Udacity applications of Term 1 and 2 run on it without any issues.
The archive with the new simulator binary is downloaded from https://github.com/udacity/self-driving-car-sim/releases/download/v1.0/term2_sim_mac.zip
It's unusually packaged - there's "term2_sim.app" directory embedded into "term2_sim_mac" - and the file access rights are not similar to the Term 1 simulator (e.g. Info.plist is not executable). The application fails to start, showing a pop-up dialog window with text "The application "term2_sim" can't be opened". If an application is packaged properly, macOS recognizes it as an app downloaded from the Internet and asks for approval to run it. However your archive doesn't behave as an application archive, but rather as a generic archive (perhaps because of the packaging issues).

awbrown90, 2017-05-06:
Sorry Yury, I am actually a windows user so I am not familiar with how Mac apps are set up. I did look at the file structure comparing Term 2 to Term 1 simulator and they looked identical as far as I could tell. @Brok-Bucholtz and @domluna were two who tested it on their local Macs.

vernor1, 2017-05-07:
The simulator app is reported to start successfully on macOS 10.11.5, but it fails on 10.12.4

vernor1, 2017-05-08:
The issue is solved by moving the app to the Applications folder. For some reason, other Udacity application can run from any download location.

closed 2017-05-08 by vernor1

jgilbert42, 2017-06-04:
Moving term2_sim.app to the /Applications folder didn't work for me when I had unzipped from the command line.

Double clicking on the zip from Finder to unzip and then right clicking on term2_sim.app and selecting open (to work around the app not being signed) worked for me.