udacity/self-driving-car-sim #30: Finding location of car with respect to the road

jahin07, 2017-05-12:
To make this simulator work with Reinforcement Learning, I need to add a reward function with respect to the car's position on the road. Is there any way the simulator can calculate the car's position?

ffrige, 2017-12-12:
Yes, there is a simple way, which is actually already implemented in the Term2 version of the simulator and used for the PID controller project. The PID observed variable is the cte (cross-track-error) and its controlled variable is the steering value.

The cte is calculated by the WaypointTracker_pid script. The calculation is based on the difference between the actual position of the car and an ideal geometrical path running around the track. The path is described by fixed "waypoints", which are already placed for you, but you can freely move them around if you want to.

Assuming you are working with the Term1 version, you need to follow these steps to be able to get the cte from the simulator:

1. copy/paste the waypoints from LakeTrackTraining scene to LakeTrackAutonomous scene

2. drop the script WaypointTracker_pid from the Term2 simulator into the Script folder

3. modify CommandServer script adding these 4 lines of code in the right place:
	private WaypointTracker_pid wpt;
	wpt = new WaypointTracker_pid ();
	var cte = wpt.CrossTrackError (_carController); //call the function to calculate cte
	data["cte"] = cte.ToString("N4"); //send cte with telemetry data

4. add this line to your drive.py script
        cte = data["cte"] # The current cross-track-error of the car

Now use your policy search algorithm to improve the policy by maximizing the reward, which is inverse proportional to the cross-track-error.

baumanab, 2018-02-19:
Thanks @ffrige!

closed 2018-02-19 by baumanab

dcompgriff, 2019-10-08:
I'm wondering if either of you have looked into writing the corresponding python code to connect with this. I'd like to take the reinforcement learning algorithms I developed for the deep reinforcement learning course and use them to train the self driving car here.

baumanab, 2019-10-08:
I don't work in the SDC program anymore, but I do think this is a great idea.