udacity/self-driving-car-sim #33: Spaming Log File

a3dbox, 2017-05-19:
The log file gets spammed with the following lines:

```
19/05/2017 21:46:37|Fatal|WebSocket.acceptException|System.Net.Sockets.SocketException: Connection refused
                            at System.Net.Sockets.Socket.Connect (System.Net.EndPoint remoteEP, Boolean requireSocketPolicy) [0x00000] in <filename unknown>:0 
                            at System.Net.Sockets.Socket.Connect (System.Net.EndPoint remoteEP) [0x00000] in <filename unknown>:0 
                            at System.Net.Sockets.TcpClient.Connect (System.Net.IPEndPoint remote_end_point) [0x00000] in <filename unknown>:0
```

In just a few seconds of running the application a log file of 100MB was generated.

This was run on Linux.  The log file can be found at : ~/.config/unity3d/Udacity/self_driving_car_nanodegree_program/Player.log

Hayoung-Kim, 2017-05-24:
> ~/.config/unity3d/Udacity/self_driving_car_nanodegree_program/Player.log

I totally agree. The log size is explosively increased so that the OS does not operate normally.

asimay, 2017-09-20:
+1, I hanged the program and forget to close, and it produced 30G log.....

asimay, 2017-09-20:
I think it should provide an option to write log or not.

`-rw-rw-r-- 1 leon leon 32482377923 Sep 20 01:08 Player.log`

xmfcx, 2017-10-27:
This log file generation physically harms my SSD. like, 9GB garbage log file written in less than 10 minutes.

tjacobs, 2017-12-04:
56 Gig logfile generated, filled up my disk. Would be nice to get this fixed. Just reduce the log frequency please.
ls ~/.config/unity3d/Udacity/self_driving_car_nanodegree_program/ -ltr
-rw-rw-r-- 1 tom tom 56806895616 Dec  4 17:27 Player.log

gittar, 2017-12-26:
This behavior did make my machine unusable in a short time (no space left on device). Could at least someone be assigned to that issue or is this problem going to stay?

asimay, 2017-12-27:
right click the execute file, and set privilege to only read, without write, will not write log.

baumanab, 2018-02-19:
Sorry for this frustration.  A note has been added to releases, letting students know about the issue and the solution.

closed 2018-02-19 by baumanab