udacity/self-driving-car-sim #35: Allowing adversaries, pedestrians and other assets

vamsiramakrishnan, 2017-05-26:
I would like to **contribute** to add adversaries and pedestrians along with assets like traffic signs in the simulator. Please suggest how I can collaborate to make myself useful.

desireend, 2017-06-06:
If possible, can you share the source code for  incorporating pedestrians and traffic system?

vamsiramakrishnan, 2017-06-06:
Dear team

     I have not yet begun to work on this but I could be of help if that's
something you are looking to add on your roadmap . Currently I am trying to
do the following .  Use Pygame to incorporate a learning mechanism where
the data generated by the car in autonomous mode is used for training the
model further, also complemented by manual data in problem areas to speeden
the training process.

    I have no experience in unity but I would like to offer my support if
required for anything specific if you may need it .

Thanks and regards
Vamsi Ramakrishnan

baumanab, 2018-02-19:
This sounds really cool.  Please let us know how this progresses.

closed 2018-02-19 by baumanab