udacity/self-driving-car-sim #36: This repository is over its data quota

keyvhinng, 2017-05-27:
Hi, I'm facing the following issue when I try to clone:

Error downloading object: Assets/1_SelfDrivingCar/Sprites/Background.png (36e4565): Smudge error: Error downloading Assets/1_SelfDrivingCar/Sprites/Background.png (36e4565319a778a53538a2c1419174cac59597859e5a8616c1b94a08d6221ac3): batch response: **This repository is over its data quota**. Purchase more data packs to restore access.

Is the repository over its allotted data quota ?

kodenusantara, 2017-05-28:
Yes I have this issue as well, I hope this get solved soon

tuzzer, 2017-05-29:
I am hitting this issue as well.

happyxujin, 2017-06-01:
I also have this issue.

drganjoo, 2017-06-01:
I am also facing this issue. Tried forking this and even then its the same message.

jwdunn1, 2017-06-03:
New month, same issue.

f1yegor, 2017-06-03:
Just to mention for future reference, that you can download prebuild one https://github.com/udacity/self-driving-car-sim#avaliable-game-builds-precompiled-builds-of-the-simulator

domluna, 2017-06-04:
Looks like Github has limits on git lfs usage https://help.github.com/articles/about-storage-and-bandwidth-usage/.

drganjoo, 2017-06-05:
@f1yegor For some reason, the prebuilt windows simulator binary (for the 2nd term) does not connect over web sockets. I downloaded the src, rebuilt it and it successfully connects. However, the blue car doesn't appear and I am assuming that is happening because I don't have the LFS files.

The prebuilt simulator that does not connect was downloaded from: https://github.com/udacity/self-driving-car-sim/releases/download/v1.4/term2_sim_windows.zip

Thanks

tuzzer, 2017-06-05:
The files sizes don't seem to be that big. Perhaps we can take them off LFS of it has data limit? 
I know that there is the precompiled version but I wanted the source so I can make changes to it. I was able to obtain a downloaded copy from someone else this time.

happyxujin, 2017-06-05:
@tuzzer Do you mind sharing the source code copy?

tuzzer, 2017-06-05:
@happyxujin Not sure how updated this was, but it worked for me.
https://1drv.ms/u/s!Ajjp7jQvYTaCjh7Rs9vmY99Vpccu

fotiDim, 2017-06-06:
Yes that's the latest. All the commits are in. Thank you

drganjoo, 2017-06-06:
@tuzzer thank you for sharing the code

kdg1230, 2017-06-17:
@tuzzer Thanx :D

Harry79, 2017-06-20:
@tuzzer Thx!

xiumingxu, 2017-08-09:
That is great! Thaks

cgarciae, 2017-09-11:
Thanks for sharing the code @tuzzer !!! 
How can this be fixed in general?

skamranali, 2017-09-18:
@tuzzer Is the downloaded copy still in the same link? The page keeps loading and shows nothing.

autonomobil, 2017-11-06:
same problem.. onedrive link doesn't work..

baumanab, 2018-02-19:
This issue has been addressed and standard cloning is functional.

closed 2018-02-19 by baumanab

AlexanderBollbach, 2018-08-26:
also having this issue

lmzh97, 2019-03-17:
@MattChanTK Not reachable for me. Can I find it someplace in github?