udacity/self-driving-car-sim #40: Release 1.4: The application “term2_sim.app” can’t be opened

cesare-montresor, 2017-06-14:
I downloaded the release 1.4, but I wasn't able to run it and I was prompt with this error "The application “term2_sim.app” can’t be opened."
For some reason it was missing the +x flag to the main binary, a quick fix is to run the following in a terminal:

`chmod +x /Applications/term2_sim.app/Contents/MacOS/term2_sim_mac`

maybe just an issue with my configuration.

laventura, 2017-06-15:
I had the same problem, with **simulator version 1.45.**

I also had to execute the same command to get it working:

```
chmod +x <Application_Dir>/term2_sim_mac/term2_sim.app/Contents/MacOS/term2_sim_mac
```

Logic4Twisted, 2017-07-15:
The same problem here with 1.45 version on Sierra version 10.12.5

mvirgo, 2018-11-19:
Closing since no further comments; often in Mac or Linux you need to add `chmod u+x` (or just +x as you have noted) for binary files to be executable.

closed 2018-11-19 by mvirgo

demigecko, 2020-01-02:
I have the same problem but it cannot be solved by previous suggestions. 
I used MacOS Catalina 10.15.2 and term2_sim version is 1.45. 
Any suggestion?