udacity/self-driving-car-sim #41: This repository is over its data quota. Purchase more data packs to restore access.

tomtran90, 2017-06-26:
I got this error while trying git clone. I'm not sure if this is from my end or not.

>git clone https://github.com/udacity/self-driving-car-sim.git
Cloning into 'self-driving-car-sim'...
remote: Counting objects: 7608, done.
remote: Total 7608 (delta 0), reused 0 (delta 0), pack-reused 7608
Receiving objects: 100% (7608/7608), 132.32 MiB | 1.40 MiB/s, done.
Resolving deltas: 100% (1987/1987), done.
Checking connectivity... done.
Downloading Assets/1_SelfDrivingCar/Sprites/Background.png (6.83 MB)
Error downloading object: Assets/1_SelfDrivingCar/Sprites/Background.png (36e4565): Smudge error: Error downloading Assets/1_SelfDrivingCar/Sprites/Background.png (36e4565319a778a53538a2c1419174cac59597859e5a8616c1b94a08d6221ac3): batch response: This repository is over its data quota. Purchase more data packs to restore access.

autodrive, 2017-07-07:
I also experience similar problem

heidihysell, 2017-07-10:
+1 on this issue

richadam, 2017-07-25:
+1 here too

byett, 2017-08-08:
Same problem here, any fix?

autonomobil, 2017-11-06:
any answer?

ilyas156, 2017-11-07:
instead of cloning download this repository as a zip and unzip it.

baumanab, 2018-02-19:
This issue has been resolved and cloning is working.

closed 2018-02-19 by baumanab