udacity/self-driving-car-sim #45: Materials and/or textures seem to be missing

papiot, 2017-07-25:
I cloned the repository using git lfs and it seems like I have all the files, however when I launch any scene, I don't see any materials and the UI looks broken. Any help much appreciated, I'd love to contribute to this project!
[image]

papiot, 2017-07-25:
You can see that the palm trees also don't have the materials showing up. Am I doing something wrong?

spencerh-b, 2017-08-29:
Are you able to start the sim?

autonomobil, 2017-11-06:
same problem..

KDragos, 2018-11-20:
I cloned the repo and opened it with Unity version 2017.4.15f1. These issues seem to be fixed.

However, in your image, it seems like the source image somehow got disconnected. To fix this:
In the Inspector window (along the right hand side), click the little circle to the right of the box next to "Source Image". This should let you choose an image to put here.
The 1_BaseImageReference should be "drivingUI_Reference". 
The 1_BaseImage should be "drivingUI_Base"

mvirgo, 2019-01-11:
No further comments; closing out.

closed 2019-01-11 by mvirgo