udacity/self-driving-car-sim #46: 50%+ CPU usage in OSX even in splash screen

antorsae, 2017-07-27:
Brings computer to a crawl, just launching the sim. Tested on OS X Sierra 10.12.5

[image]

tjacobs, 2017-07-28:
Seconded, I had to shut down the app between each debug, which slowed the project development

jgilbert42, 2017-07-29:
This isn't an interactive performance problem for me, but it kills the battery pretty quick.

wAuner, 2017-11-28:
The same is true for linux.
I use a Nvidia GTX 1080 and even in idle mode I get 20-25% GPU usage as well as 20-25% CPU usage.
This should really be fixed.

kerys, 2017-12-08:
I second this, I have > 80% CPU usage in splash screen with Ubuntu 16.04 as well. There appears to be a bug with the simulator implementation. 
[image]

roman-smirnov, 2018-11-26:
At first I thought it might be a hardware acceleration issue, but I've confirmed it does run on the discrete GPU. Moreover, it looks like CPU usage doesn't change much between the lowest and highest visual quality settings and resolution, so the issue is likely unrelated to graphics. 

I've also looked into CPU and memory usage over time thinking it might be resource leak. It looks like there's no noticeable performance degradation during prolonged usage and multiple runs.

It seems like either the implementation itself is very CPU heavy for some reason or there's something causing high CPU usage on macOS. 

If someone wants to dig a little deeper into the issue (and is familiar with Unity), a good next step is to set up the project and run a CPU profiler to pinpoint the issue.

__Resource usage shortly after startup (lowest settings)__:
[image]
[image]

__Resource usage after some time and several runs (lowest settings)__:
[image]
[image]

__Resource usage after a few runs at highest settings__:
[image]
[image]

tsigalko18, 2020-07-20:
I've used and tested the simulator on both Windows and MacOS Mojave. When executed on the Mac, the simulator seems to suffer from a performance bug which seems to be due to [Unity v.5.5.6f1](https://forum.unity.com/threads/unaccounted-time-between-start-of-frame-and-camera-render.444095/). Basically, 10-30% of CPU is drained by this bug. It seems to be solved in the latest Unity versions, but compatibility of the simulator is still an open issue.

[image]