udacity/self-driving-car-sim #47: Kidnapped vehicle control data seems to have an error at line 238

khushn, 2017-07-28:
Hi!
I noticed that I was getting yaw_rate input values of 62.697(!) around step 250. Because of which yaw error always remains above the threshold. Even though the particle filter works. 

I looked up the files and notice that in version 1.45 the file Assets/Resources/observations/control_data.txt, has the following entry at line 238: 

8.9471 62.697

Which seems much higher than all the other values. Is this an error?

alistairskirk, 2017-08-01:
I am also having this issue. My particle filter works everywhere except when this 62.697 yaw_rate shows up from the controls, it increases all future yaw_rate errors.

swwelch, 2017-08-04:
Checked with @awbrown90 - It looks like this is in the right range mod 2pi. If you normalize the incoming angles, does it work for you?

khushn, 2017-08-05:
Honestly, there is some confusion regarding this value. It will be great if the devs can clarify. This is yawrate so taking a mod upfront, can change its meaning. So we should do it after multiplying with delta_t. Right? 
I tried two or three combinations. Eventually what worked is what is described in this thread: 
https://discussions.udacity.com/t/beware-simulator-expects-yaw-in-0-to-2-pi-range-not-pi-to-pi/295952

But, I'd still hope if devs correct (or explain) this value. As its well outside the normal curve of yawrate values.

alistairskirk, 2017-08-08:
> expects-yaw-in-0-to-2-pi-range-not-pi-to-pi/295952

This was the issue. I was normalizing to -pi to +pi, and the simulator is
expecting theta to be 0:2*pi.
Adding the two lines to main.cpp as referenced in that link fixed my yaw
error.

closed 2018-02-19 by baumanab