udacity/self-driving-car-sim #50: White layer over simulator problem

mattk90, 2017-08-11:
Hey people,

I'm experiencing some issue with my simulator. The background of the UI is white in the menu and also during the simulator there is a white layover. I though it might have something to do with the Build Settings and already tried if I need to include maybe UISetupScene from the assets but it didn't help. Any ideas or did some of you experience the same problem? I'm quite a beginner with Unity so far, so sorry if that's an easy question, though couldn't find anything online.

Screenshots are attached:

[image]
[image]
[image]

DevParapalli, 2017-09-29:
Same here.....BTW which version of Unity are you using?

autonomobil, 2017-11-06:
same problem..

salal40, 2017-12-14:
@mattk90 the reason for that white overlay is the missing textures. When you download the archive from Github using "Download ZIP", the texture files which are stored on git using Git LFS(Large File Storage), are not downloaded. Instead, only the 'pointers' to those files are downloaded. So in order to solve the issue, either install LFS in git and then clone the repo, or use some other git client which supports LFS.
You can find the info here as well:
https://help.github.com/articles/versioning-large-files/

closed 2018-02-19 by baumanab