udacity/self-driving-car-sim #53: Term 3 integration: unpredictable reactions to throttle\brake commands

amakurin, 2017-08-27:
I've collected a list of issues with throttle\brake commands of integration project simulator. Issues were reported by me and other teams\students.

[The car doesn't move even if throttle=100 is published in /vehicle/throttle_cmd](https://carnd.slack.com/files/gpavlov/F6U53K2CU/Car_is_not_moving.txt)

[Sequence like throttle 5. throttle 5. brake 0. brake 0. leads to car is not moving](https://carnd.slack.com/archives/C6NVDVAQ3/p1503303623000074)

[If i send eg throttle 0.2 only once And then send throttle 0.0 once or at some frequency, whatever, car is still increasing speed.](https://carnd.slack.com/archives/C6NVDVAQ3/p1503409009000404?thread_ts=1503303623.000074&cid=C6NVDVAQ3)

[I’m seeing very weird behaviour when trying to make the car brake. To come to full stop from speed of ~20mph car needs about 100metres  and 20 sec.](https://carnd.slack.com/archives/C6NVDVAQ3/p1503732084000029)

[My car is just not moving VirtualBox + Mac. One out of 100 times it does move but it drives straight of the road.](https://carnd.slack.com/archives/C6NVDVAQ3/p1503819283000011)

PuchatekwSzortach, 2017-08-29:
@awbrown90 I believe all issues stem from two features of the simulator:
- it doesn't continuously apply input until new one arrives, but only applies it momentarily and then goes to neutral until new input arrives
- it doesn't process only most recent input, but holds a queue of all inputs, leading to very erratic behaviour when network delays make inputs queue fill up

Here's a description of sample issues I am seeing, copied from slack:

Around waypoints 6400~6600 some of the runs car first slows down a bit for a few seconds then start accelerating by itself despite controller quickly reacting and applying 0 throttle and max brake. What I suspect is happening is that some communication delay blocks all throttle and steering commands coming into simulator and once delay goes away all queued up commands are being applied to simulator in very short successions. @awbrown90 - can you please confirm or deny whether simulator continues to apply inputs until new inputs arrive? Because it really looks like it doesn’t and instead applies them only momentarily and at pace at which they arrive - leading to behaviour I’m running into.

I have logs showing target velocities are constant and sensible, waypoints are smooth and sensible, errors driving my steering and throttle controllers are correct and inputs to car are sensible too. Simulator is simply not responding to inputs in real time. Quite sure you are having a problem with 10 secs worth of old queued up inputs being applied in like 1 sec. By the time fresh inputs that reacts to these changes get to have a saying, car is already off the road.

gpavlov2016, 2017-08-29:
I am not able to make the car brake at all, even if I stop all throttle commands and apply max brake - the car continues to accelerate.
Do you think it is related somehow?

kuba-lilz, 2017-08-29:
If car reacts to steering commands while ignoring brake, then likely not. But I hear about many people having same problem as you do on Slack. For us braking "sort of works", but takes huuuge amount of time and distance - something of the order of 200 metres and 15sec to brake from 20mph to stop.

gpavlov2016, 2017-08-29:
Turns out the steering is not working either, and the speed keeps increasing even though I publish throttle=0 so looks like the throttle command is ignored as well.
I checked the queue theory and tried to run the server at 1Hz, 10Hz, 50Hz, 100Hz, 200Hz and saw absolutely no difference in behavior between all of these.

kuba-lilz, 2017-08-29:
@gpavlov2016 In that case are you sure your ROS is actually connecting to simulator?

amakurin, 2017-08-29:
By the way live help said that this could be caused by lack of RAM or CPU or GPU. Here is my setup macOS Sierra 2.5GHz Core i7, 16GB, Nvidia GeForce GT 750M, SSD. Used to run both simulator and VM - udacity provided image over Virtualbox.

kuba-lilz, 2017-08-29:
I'm using 2017 Macbook Pro with 3.1 GHz Intel Core i7, 16 GB 2133 MHz LPDDR3 and Radeon Pro 560 4096 MB, giving full resources to VM. If this isn't enough, what is?

mirouxbt, 2017-08-29:
Having the same issue with 3.1 GHz Intel Core i3, 16GB, Geforce GTX 1060. I am not using the VM at all, but still get those nasty lag which make really difficult to get consistent result or any result at all.

jdleesmiller, 2017-09-09:
I found some workarounds on Slack that helped work around these issues, so I think it's worth posting them here. The changes in this commit improved performance a lot:

https://github.com/jdleesmiller/CarND-Capstone/commit/33dae9248a73feab9b577dd135116b6575e85788

jwdunn1, 2017-09-25:
At waypoint 6665, the images being sent to the controller double in data length. This causes server.py to max CPU utilization (even if all sio events are commented) and extreme FPS drop in both the controller and the simulator. Also occurs at the beginning of track near the speed limit signs. In both these areas, the additional trees lead to complexity in the images transmitted. To prevent this bottleneck, the scene needs to be simplified **or** the image size needs to be capped.