udacity/self-driving-car-sim #54: Filling ~/.config/unity3d/Udacity/self_driving_car_nanodegree_program/Player.log if connection fails

HTuennermann, 2017-09-05:
The simulator generates a lot of log entrys like this if ros is not running int the Player.log file:

9/5/2017 1:59:25 AM|Fatal|WebSocket.acceptException|System.Net.Sockets.SocketException: Connection refused
                            at System.Net.Sockets.Socket.Connect (System.Net.EndPoint remoteEP, Boolean requireSocketPolicy) [0x00000] in <filename unknown>:0
                            at System.Net.Sockets.Socket.Connect (System.Net.EndPoint remoteEP) [0x00000] in <filename unknown>:0
                            at System.Net.Sockets.TcpClient.Connect (System.Net.IPEndPoint remote_end_point) [0x00000] in <filename unknown>:0

after one night leaving the program on, my Player log was 180GB.

antorsae, 2017-10-26:
I can confirm and reproduce in Ubuntu 16.04. Emptied my HD space in the process.

mreichelt, 2017-10-26:
@HTuennermann @antorsae until this issue is fixed I have a quick & nice workaround for you, I also posted it in [Slack](https://carnd.slack.com/files/U46BXV3FZ/F7FV13HLH/prevent_player_log_from_being_created.sh):

```sh
# on Linux
rm ~/.config/unity3d/Udacity/self_driving_car_nanodegree_program/Player.log
mkdir -p ~/.config/unity3d/Udacity/self_driving_car_nanodegree_program/Player.log

# on Mac
rm ~/Library/Logs/Unity/Player.log
mkdir -p ~/Library/Logs/Unity/Player.log
```

Hope it helps!