udacity/self-driving-car-sim #55: Udacity Simulator does NOT compile in Unity

happyguard91, 2017-09-24:
I downloaded your simulator source code from your github, unzipped the folder, and then loaded this folder to Unity. When Unity opens after loading and compiling scripts, it returns 218 errors most of them saying that certain files cannot be read. 

Please update your current source code.

autonomobil, 2017-11-06:
same problem

ffrige, 2017-12-12:
You cannot just "download" the simulator because it is stored with Git LFS (large file storage), which means that most large files (mainly images) are replaced by simple text pointers.
What you need to do is:
1. download and install the LFS extension to git from https://git-lfs.github.com/
2. type "git lfs install" in your command window
3. now use standard git to clone the repository, i.e. type "git https://github.com/udacity/self-driving-car-sim.git" in your command window from the folder where you want to save the data. Note that it will take a while, it is over 1GB!

mvirgo, 2018-03-15:
Looks like this is solved since no recent comments; the above need to install git lfs is also included in the instructions.

closed 2018-03-15 by mvirgo