udacity/self-driving-car-sim #59: Linux: Simulator only works if system language is set to English

wAuner, 2017-11-14:
When using a language other than English for your linux system (tested with German) the simulator does not work properly. 
[image]
The result is a simulator with no visible datapoints and the udacity logo on the car is jumping around. 
If you connect a working (correctly implemented) Filter, the RMSE explodes.
I guess this has something to do with the way the data is read. Probably due to some formatting error the datapoints don't appear on the screen if the system language is not English.
Tested this with KDE Neon, Linux Mint 18.2 and Ubuntu 16.04 LTS.

Switching the system language in the preferences to English doesn't solve the problem, but the simulator works correctly.

mreichelt, 2017-11-14:
Should be closed, duplicate of #6

wAuner, 2017-11-14:
you‘re right it‘s a duplicate. however this title could help others with that problem, since this problem isn‘t described in the lessons.

baumanab, 2018-02-19:
Sorry for this issue.  A note has been added to the simulator release notes, regarding the issue, and its solution.

closed 2018-02-19 by baumanab