udacity/self-driving-car-sim #71: Is it able to get image in real time?

junghs0226, 2018-02-11:
Hi. 
Now, I have a project about autonomous driving by using image as input.
So I need a image in real time. I finished setting up.
Is there any method to get image from uda car sim?
or does it support to do?
thanks.

karray, 2018-02-28:
I'm also looking for method to capture an image form a camera.
The capture is realized in the [CameraHelper.cs](https://github.com/udacity/self-driving-car-sim/blob/master/Assets/1_SelfDrivingCar/Scripts/CameraHelper.cs) and it is used in [CommandServer.cs](https://github.com/udacity/self-driving-car-sim/blob/master/Assets/1_SelfDrivingCar/Scripts/CommandServer.cs)
I applied the same method in my own project, but it freezes the scene

mvirgo, 2018-03-15:
Check out the repository for Behavioral Cloning (the third project in Term 1 of the Self-Driving Car Engineer Nanodegree program) [here](https://github.com/udacity/CarND-Behavioral-Cloning-P3) - the `drive.py` file contains code that pulls image in real-time.

closed 2018-03-15 by mvirgo