udacity/self-driving-car-sim #72: Simulator crashes on Linux Mint 18 (Ubuntu 16

rafi-kamal, 2018-02-19:


xwzy, 2018-03-13:
I have the same issue.

```
Set current directory to /home/wzy/Desktop/term2_sim_linux
Found path: /home/wzy/Desktop/term2_sim_linux/term2_sim.x86_64
Mono path[0] = '/home/wzy/Desktop/term2_sim_linux/term2_sim_Data/Managed'
Mono path[1] = '/home/wzy/Desktop/term2_sim_linux/term2_sim_Data/Mono'
Mono config path = '/home/wzy/Desktop/term2_sim_linux/term2_sim_Data/Mono/etc'
displaymanager : xrandr version warning. 1.5
client has 16 screens
displaymanager screen (0)(Virtual1): 1440 x 900
Using libudev for joystick management

Importing game controller configs
Aborted (core dumped)
```

soaresinfo, 2018-03-13:
Mint (18.3) user here, I just double click the executable (beta_simulator.x86_64) with the lowest res 640x480 on my notebook, it run ok.

xwzy, 2018-03-14:
I use a fresh ubuntu 16.04, and I still crashed after choose 640x480 or any other config.
Can anyone help me?

```
$ ./term2_sim.x86_64 
Set current directory to /home/wzy/Desktop/term2_sim_linux
Found path: /home/wzy/Desktop/term2_sim_linux/term2_sim.x86_64
Mono path[0] = '/home/wzy/Desktop/term2_sim_linux/term2_sim_Data/Managed'
Mono path[1] = '/home/wzy/Desktop/term2_sim_linux/term2_sim_Data/Mono'
Mono config path = '/home/wzy/Desktop/term2_sim_linux/term2_sim_Data/Mono/etc'
displaymanager : xrandr version warning. 1.5
client has 16 screens
displaymanager screen (0)(Virtual1): 1440 x 900
Using libudev for joystick management

Importing game controller configs
Aborted (core dumped)
```

xwzy, 2018-03-14:
I installed ubuntu on a virtual machine on macbook

rafi-kamal, 2018-03-25:
I downloaded the [v1.45](https://github.com/udacity/self-driving-car-sim/releases/tag/v1.45) again, this time it worked. I think I have been using v1.3 before.

xwzy, 2018-04-10:
@rafi-kamal No, v1.45 still crash under virtual machine on my mac. But a ubuntu on PC is ok to run.

```
wzy@wzy:~/Desktop/term2_sim_linux$ ./term2_sim.x86_64 
Set current directory to /home/wzy/Desktop/term2_sim_linux
Found path: /home/wzy/Desktop/term2_sim_linux/term2_sim.x86_64
Mono path[0] = '/home/wzy/Desktop/term2_sim_linux/term2_sim_Data/Managed'
Mono path[1] = '/home/wzy/Desktop/term2_sim_linux/term2_sim_Data/Mono'
Mono config path = '/home/wzy/Desktop/term2_sim_linux/term2_sim_Data/Mono/etc'
displaymanager : xrandr version warning. 1.5
client has 16 screens
displaymanager screen (0)(Virtual1): 1440 x 900
Using libudev for joystick management

Importing game controller configs
Aborted (core dumped)

```

tsbertalan, 2018-04-18:
It crashes for me without the `Aborted (core dumped)`. I get

```
Set current directory to /home/tsbertalan/usr/term2_sim_linux
Found path: /home/tsbertalan/usr/term2_sim_linux/term2_sim.x86_64
Mono path[0] = '/home/tsbertalan/usr/term2_sim_linux/term2_sim_Data/Managed'
Mono path[1] = '/home/tsbertalan/usr/term2_sim_linux/term2_sim_Data/Mono'
Mono config path = '/home/tsbertalan/usr/term2_sim_linux/term2_sim_Data/Mono/etc'
displaymanager : xrandr version warning. 1.5
client has 2 screens
displaymanager screen (0)(DVI-I-2): 1920 x 1080
displaymanager screen (1)(DVI-I-3): 1080 x 1920
Using libudev for joystick management

Preloaded type GtkListStore
Preloaded type GtkWindow
Preloaded type GtkVBox
Preloaded type GtkImage
Preloaded type GtkNotebook
Preloaded type GtkHBox
Preloaded type GtkFrame
Preloaded type GtkAlignment
Preloaded type GtkTreeView
Preloaded type GtkLabel
Preloaded type GtkCheckButton
Preloaded type GtkScrolledWindow
Preloaded type GtkComboBox
```

I then choose a resolution, graphics quality, and window mode (all give same results) and it exits with no further output.

tsbertalan, 2018-04-18:
I also tried restoring my write permission to `~/.config/unity3d/Udacity/self_driving_car_nanodegree_program/Player.log`, but that didn't help. I'm on Ubuntu 16.04.04 LTS

tsbertalan, 2018-04-18:
Aaand it's gone after reboot. I believe it was my recent reinstall of the Nvidia 390.48 driver. So, I guess, one for the knowledgebase.

kk2491, 2019-01-09:
yea... @tsbertalan you are right,.. After reinstalling the NVIDIA driver simulator works perfectly fine.

mvirgo, 2019-01-11:
Looks like from above this issue is resolved / user side; closing.

closed 2019-01-11 by mvirgo

rijo11811, 2021-06-08:
Faced the same issue today. Thought its because of my laptop driver. 
Solution : Untick Windowed option. Works like butter.