udacity/self-driving-car-sim #85: Term 2 Simulator crashes on macOS

kkarnatak, 2018-06-27:
Hi,

I am trying to run "Term 2 Simulator v1.45" on MacOS high sierra. The app crashes every time. I have also tried the lower version of the simulator, but no luck.

Any suggestions?

Thank you.

Anner-deJong, 2018-07-26:
Just to inform:
Downloaded term2 v1.45 and ran it successfully on High Sierra.
I tried with highest screen resolution and highest graphics.

However, the application seems to be using up a lot of my CPU. With medium screen resolution and highest graphics my build-in fan started making considerable noise already.

I suggest you try with lowest settings as well.
Furthermore, I assume it requires some c++ compiler and perhaps other dependencies on your computer (havent read through the project yet though). Are you aware of anything that might not be set up correctly on your local mac?

mvirgo, 2019-01-11:
No further recent comments, closing.

closed 2019-01-11 by mvirgo