udacity/self-driving-car-sim #90: slef-driving-car-sim - term2_sim.x86_64 - not running on ubuntu 16

kzoheir, 2018-08-14:
I've downloaded the file: term2_sim_linux.zip, extracted it, then changed permission for the file: term2_sim.x86_64 to be executable.

When running term2_sim.x86_64 it asks for resolution. Then nothing happened. I was expecting to have a screen with a menu to select: "Bicycle tracker with EKF."

Are there any prerequisites? Any missing step in the middle?

doYrobot, 2018-11-02:
I had the same problem.

doYrobot, 2018-11-02:
I found that the file did not have permission to run.

chmod ug+x ./term2_sim_linux/term2_sim.x86_64

kzoheir, 2018-11-08:
Thank you.

My problem was different. It's more related to Ubuntu installation! It only
works when I start Ubuntu in a safe mode and it's slow.

sie sie
Khaled

mvirgo, 2018-11-20:
Looks like you have solved the issue, so closing this out.

closed 2018-11-20 by mvirgo