udacity/self-driving-car-sim #94: Is it possible to use the 3'th party's Vehicle dynamics model

s894330, 2018-11-15:
Hi ,

As the title say , Could I use 3'rd party Vehicle dynamics model (like CarSim) to do simulation ?
If the answer is "No" for now , would you have the plan to add this feature in the feature?

Thank you

mvirgo, 2018-11-19:
As long as you are sending data to the correct port (typically 4567 for these simulators) and in the correct format, they can communicate with each other. It would be up to you to write the code necessary to make sure this is done (see [Behavioral Cloning](https://github.com/udacity/CarND-Behavioral-Cloning-P3/blob/master/drive.py) for an example in Python or [Extended Kalman Filter](https://github.com/udacity/CarND-Extended-Kalman-Filter-Project/blob/master/src/main.cpp) for an example in C++).

mvirgo, 2018-12-07:
Closing out since no further comments in 2+ weeks.

closed 2018-12-07 by mvirgo