udacity/self-driving-car-sim #95: How to get the road structure or road network from Simulator Term3-1.2V?

wenjiebit, 2018-11-28:
Hi, Thanks for your great job! If I want to get the road network from the simulator, what should I do? I didn't find this kind of data in json data of main.cpp. Thanks !

mvirgo, 2018-11-30:
Hi there! What exactly do you mean here by the "road network"? Waypoints, the map, etc.?

The "map" is contained within the data folder - `highway_map.csv` Note that from main.cpp we can see this is made of x, y, s, d_x and d_y values.

Separately, you may have already been doing this, but you can see all the other possible information coming from the simulator by just doing `cout << j[1] << endl;`; there is no extra info contained therein that the simulator is currently configured to send out, that is not already being gathered in main.cpp.

If there is additional data past this you want to gather, you'd want to delve into the Term 3 simulator code contained in this repository, as you'd need to compile the simulator on your own through Unity to produce additional information.

mvirgo, 2019-01-11:
No further comments; closing.

closed 2019-01-11 by mvirgo