udacity/self-driving-car-sim #96: Unable to make connection between drive.py and simulator

canozcivelek, 2018-11-29:
Hi! I'm having some trouble establishing connection between drive.py and simulator. Normally, when I run the code, it should successfully connect and send throttle/steering commands so the car drives. But nothing happens. Attached are the screenshots. Can someone help me? What am I missing here?
[image]

andrepith, 2018-11-30:
I've got this problem too, i still try to find the solution

canozcivelek, 2018-11-30:
Hi, thanks for the tip. Can you tell me how to downgrade it? Thanks!

vishal-noob, 2018-11-30:
Downgrading works like a charm. Thank you @andrepith  <3

canozcivelek, 2018-11-30:
Yess it does! I can't thank you enough. Thank you!

vishal-noob, 2018-11-30:
Where to change the directory for loading the new csv file?

Pranav5603, 2021-04-09:
Error after installing engineio 2.2.0:
AttributeError: module 'engineio' has no attribute 'WSGIApp'
Please help!