udacity/self-driving-car-sim #97: "GET /socket.io/?EIO=4&transport=websocket HTTP/1.1" 404

matthewgan, 2018-12-07:
I started the builted package of the latest release Unity exe
Try to connect to the socket server, but i got this error all over looping on my terminal
"GET /socket.io/?EIO=4&transport=websocket HTTP/1.1" 404

Please kindly help to tell me what am i be wrong of connect to the server?
I am connect to localhost:4567

Hema91, 2018-12-08:
i have the same issue in both:
- native ubuntu installation with ros and the simulator
- virtual image on win 10 with the simulator running on either windows and inside the vm

trn84, 2018-12-10:
I am having the same issue

mvirgo, 2018-12-10:
Hi all - which simulator is it you are getting this issue for?

Hema91, 2018-12-11:
System Integration v1.3
https://github.com/udacity/CarND-Capstone/releases

trn84, 2018-12-11:
Hi,

yes, with both, v1.3 and the older one....There is no issue when running within the VM...

Hema91, 2018-12-11:
i re-installed everything and it worked with the VM on win 10 but it dosn't last long, after about 1~2 minutes the connection stops and the msg "GET /socket.io/?EIO=4&transport=websocket HTTP/1.1" appears

lukashubl, 2018-12-12:
got the error today as well ...

lukashubl, 2018-12-12:
found the solution here from Onixaz: https://github.com/llSourcell/How_to_simulate_a_self_driving_car/issues/34#issuecomment-445077710

matthewgan, 2018-12-12:
That issue is also from me. But the solution is not working for me.

trn84, 2018-12-12:
thank you @lukashubl I have uninstalled python-socketio and installed the newest version. Now it works :-)

matthewgan, 2018-12-20:
update python-socketio to latest and it finally works

gpokhark, 2019-02-10:
I am having same issue on my local drive virtual machine and  I am not able to resolve it. I tried updating the python-socket it didn't work. I would appreciate if someone could help. It was working on the Udacity work-space but when I tried one day later it gave me same error. I didn't update anything on the Udacity work-space. I have python-socketio==1.6.1 on the work-space and it was working fine until yesterday.