udacity/self-driving-car-sim #100: can't download "Avaliable Game Builds"

daohu527, 2018-12-22:
I can't download Avaliable Game Builds from "https://d17h27t6h515a5.cloudfront.net/topher/2017/February/58983558_beta-simulator-linux/beta-simulator-linux.zip". Is the link is broken ,or is it my network problem?

mvirgo, 2018-12-22:
Hi there! The link still works, although I have heard that cloudfront gets blocked in China, so if you are attempting to access it from there it may be part of the issue. I will try to see if I can find any alternative links we may have provided.

daohu527, 2019-01-05:
> I think otherwise if you try the [Release page](https://github.com/udacity/self-driving-car-sim/releases) you should be able to download it.

It seems works.Thx! By the way, could you pls update the readme,then otherone have the same problem can see.

mvirgo, 2019-01-11:
Readme is now updated for V2 of Term 1, so closing out.

closed 2019-01-11 by mvirgo