udacity/self-driving-car-sim #104: Assets/Standard Assets/Cameras/Scripts/TargetFieldOfView.cs(62,53): error CS0246: The type or namespace name 'ParticleRenderer' could not be found (are you missing a using directive or an assembly reference?)

raoping2017, 2019-01-26:
Assets/Standard Assets/Cameras/Scripts/TargetFieldOfView.cs(62,53): error CS0246: The type or namespace name 'ParticleRenderer' could not be found (are you missing a using directive or an assembly reference?)

ICEBERG-VR, 2019-02-15:
Same pb here after migrating a project from 2018.2 to 2018.3

YiheChen1995, 2019-02-22:
Same problem. Any Idea how this happened?

daxiao, 2019-02-28:
Same problem

xooxo, 2019-03-18:
It is because Unity removed all functionality of API regarding ParticleRenderer.

Stormwaker, 2019-05-08:
Just use Unity 2018.2 for this project. Anything newer will cause this error to appear.

marianpekar, 2019-06-13:
As @xooxo said, ParticleRenderer has been removed. ParticleSystem is need to be used instead.
https://docs.unity3d.com/ScriptReference/ParticleSystem.html

marianpekar, 2019-06-13:
As @xooxo said, ParticleRenderer has been removed. ParticleSystemRenderer is need to be used instead.
https://docs.unity3d.com/ScriptReference/ParticleSystemRenderer.html

MrGT07, 2022-01-26:
Just remove that part of the code from the OR condition and it would still works the same as the ParticleRenderer returns "true" anyway.