udacity/self-driving-car-sim #106: GET /socket.io/?EIO=4&transport=websocket HTTP/1.1" 200 0 45.359629

gpokhark, 2019-02-11:
I am having same issue on my local drive virtual machine and I am not able to resolve it. I tried updating the python-socket it didn't work. I would appreciate if someone could help. It was working on the Udacity work-space but when I tried one day later it gave me same error. I didn't update anything on the Udacity work-space. I have python-socketio==1.6.1 on the work-space and it was working fine until yesterday.

I am getting following issue on the local virtual machine as well as on Udacity's work-space and I am not able to sort it out, help will be really appreciated.

"GET /socket.io/?EIO=4&transport=websocket HTTP/1.1" 200 0 45.359629"

liupaladin, 2019-06-01:
Have you figured out? I have same issue

gpokhark, 2019-07-30:
@liupaladin I think yes I have it happens when your memory is full. To solve this issue I would suggest using a high end computer with high GPU and RAM and or the provided workspace.

closed 2019-07-30 by gpokhark