udacity/self-driving-car-sim #116: Very basic! But please help, can not save training data

hbzhang, 2019-10-06:
[image]

There, I try to record data on my Mac, I can not pass the step of choosing a folder to save data. Why?

supermomie, 2019-10-06:
Hi, u need to select a correct folder in your path

hbzhang, 2019-10-06:
What do you mean correct folder? Tanks!

kvijaykiran, 2020-02-22:
I had the same issue. After a bit of scratch my head, I figured it out. It was a duh moment.
You need to go one step up, to /Users/zhangh. Then, select 'drone', with your mouse, but do not double click. Then, hit the 'Select' button. That should do it.