udacity/self-driving-car-sim #118: Support for macOS Catalina (v10.15)

jinay1991, 2019-10-12:
With current release binaries for all the simulators, It does not support macOS Catalina.

When launched, following error is coming: 

```
LSOpenURLsWithRole() failed with error -10810 for the file /Users/.../term2_sim_mac/term2_sim.app.
```

Same is for all the other simulators. 

Request to add prebuilt binaries for macOS Catalina (v10.15).

endymioncheung, 2019-10-20:
@jinay1991 , currently I am on Mac OS Catalina v10.15 too. Like you said most Mac release does not work..I found that the Term 1 Version 1 works on my Mac. This link below is copied from the README on this repo.

https://d17h27t6h515a5.cloudfront.net/topher/2016/November/5831f290_simulator-macos/simulator-macos.zip

jinay1991, 2019-10-20:
@endymioncheung Thank for your reply, though I request to have newly compiled binaries for Term2/3 with support of MacOS Catalina as they provide more functionality than of term-1.

endymioncheung, 2019-10-20:
@jinay1991 I have got some success compiled the Udacity vehicle simulator for the Mac OS Catalina on my MacBook Pro this morning after some minor tweaks. You can download it here: the https://github.com/endymioncheung/Udacity_vehicle_sim_MacCatalina.git

jinay1991, 2019-10-20:
> @jinay1991 I have got some success compiled the Udacity vehicle simulator for the Mac OS Catalina on my MacBook Pro this morning after some minor tweaks. You can download it here: the https://github.com/endymioncheung/Udacity_vehicle_sim_MacCatalina.git

How do you unzip files? I can not run app, I get following error.

```
$ open myUdacitySim.app
The application cannot be opened because its executable is missing.
```

jinay1991, 2019-10-20:
Thanks @endymioncheung for your support. Although I am still not able to launch the simulator with macOS Catalina. When I launch, it launches with out error but flash and terminated immediately. Was that same observation for you too?

endymioncheung, 2019-10-20:
I checked it again just then and manually download the "ZIP" from my Github link I mentioned above. Then unzip it by double clicking the "myUdacitySim.z01", I noted that double clicking the "myUdacitySim.zip" won't work because it is a multi segment file because Github does not allow file larger than 100Mb.

I used the unzip tool downloaded from "https://theunarchiver.com"..

Also after you unzipped successfully, the system by default will block it and prevent it to run as the "Author" is unknown and cannot be verified, so you have to manually authorize to open this file. 

Usually I don't upload executable on Github as I think its dangerous to run unknown app, I built this app originally for myself. Just thought to share it others and save others hours of pain. Just a small disclaimer, I compiled this code on my computer and assumed have no computer virus, and I take no responsibilities for any data loss or damages. Use it at your own risks :)

If you are okay with the disclaimer, then go to "System Preferences\Security & Privacy", click the General tab, then on the right bottom corner, click "Open Anyway".

Have fun!

jinay1991, 2019-10-25:
Thank you very much @endymioncheung for providing Term2 Simulator. Though as I mentioned earlier, I need term3 (branch: term3_collection) simulator. If you have env setup, can you please build the code for term3 and share the binary?

jinay1991, 2019-10-26:
Finally was able to build Term3 Simulator on my MacOS. Here is the binary running just fine for macOS Catalina. 

https://github.com/jinay1991/motion_planning/releases/download/v1.1/term3_sim_mac_catalina.zip

closed 2019-10-26 by jinay1991

endymioncheung, 2019-10-26:
@jinay1991, just saw your post and update. So you have got the Term 3 simulator to work?

I took a look at the link you posted https://github.com/jinay1991/motion_planning/releases/download/v1.1/term3_sim_mac_catalina.zip

The link seems to be removed. Can you upload the link again, buddy? Thanks

jinay1991, 2019-10-26:
https://github.com/jinay1991/motion_planning/releases/download/v1.1/term3_sim_mac_catalina.zip

endymioncheung, 2019-10-26:
It works. Thanks for sharing the binary of the Term 3 simulator :)

taytay0904, 2020-03-19:
Hey, I wonder how did you guys build successfully. Can I know what version of Unity are you using? I tried switch back to 5.5.1 the app load infinitely. If I use the newest version, like 2019, it contains build error such as ParticleRenderer' could not be found (are you missing a using directive or an assembly reference?)

endymioncheung, 2020-03-19:
I have seen this error before. I cannot remember exactly how I resolved it
at the end, I think I just simply removed that keyword, it seems okay to
build and I was able to use it for the term 1 and term 2 SDCND project.

I also found a link here about the ParticleRenderer issue:
https://forum.unity.com/threads/there-is-no-particlesystem-in-c-script-when-i-disable-the-wind-module.562072/

I guess you are using Mac OS Catalina too, I can understand your
frustration. I can upload a binary I built for my mac to my Github in a few
hours time later at https://github.com/endymioncheung. I will label the
repo name something starts with SDCND_MacCatalinaSimulator. Please "star"
that repo if you like it ;-)

Have fun Udacitying.

Endy

endymioncheung, 2020-03-20:
Sorry for the late upload taytay0904 , I was having problem splitting the
zip files before.

Here is the updated repo
https://github.com/endymioncheung/CarND-MacCatalinaSimulator, I've tested
the unzip using Mac Terminal on Mac OS Catalina 10.15.3.

taytay0904, 2020-03-21:
hey thank you very much. I will try it later today

praveenbommalibits, 2020-07-23:
Thanks a lot @endymioncheung @jinay1991