udacity/self-driving-car-sim #128: Simulator not compatible with newer versions of Linux OS

gabesj, 2020-10-08:
The Linux Term 2 Simulator v1.45 program crashes after clicking Ok in the dialogue box where the resolution and quality are selected.  It crashes about the time that the "Made with Unity" splash screen should appear and before the screen that allows you to select which simulator to run.  It seems to be independent of my project code as this happens before the websocket connection is made.  STRACE says that it crashes on a SIGABRT fault: `+++ killed by SIGABRT (core dumped) +++ Aborted`   I had this issue on a computer running Fedora 32, on a different computer running Ubuntu 20.04.1, and also on a third computer running a VMWare virtual machine of Ubuntu 20.04.1.  When I tried it on the third computer in a virtual machine running Ubuntu 16.04.7 then it worked as expected.  It also worked in a virtual machine running Ubuntu 18.04.7.  I wish I had more details other than a correlation between OS version and the fault, but I hope that gives a good starting point to dig deeper.

RifkyYP, 2020-10-20:
@gabesj do u already have a solution for this problem? i also getting the same issue when running the Term3 Simulator v3.12 in my Ubuntu 20.04.1 and Linux Mint 20 CInnamon

gabesj, 2020-10-25:
@RifkyYP Sorry, I haven't found a solution.  I ended up installing Ubuntu 18 on a second computer and used that computer to run the simulator.

RifkyYP, 2020-10-25:
okay, thank you for your response

seongyooncho, 2021-05-13:
I have the same problem. A simple workaround I found is running in fullscreen by unchecking **windowed**. It is a bit disturbing. But at least it works without a problem.

EthanCodesss, 2021-05-17:
Unchecking **windowed**. It also works for me.