udacity/self-driving-car-sim #131: Server code actively refuses to connect with Udacity's Self Driving Car Simulator (Port 4567)

thetechdude124, 2021-01-08:
When running the drive.py and simulator files, the connection is never established - it simply says "accepted", rather than actually connecting. When I looked at the output log for the simulator, here's what I found:

`|Fatal|WebSocket.acceptException|System.Net.Sockets.SocketException: No connection could be made because the target machine actively refused it.
                      
                        at System.Net.Sockets.Socket.Connect (System.Net.EndPoint remoteEP, Boolean requireSocketPolicy) [0x00000] in <filename unknown>:0 
                        at System.Net.Sockets.Socket.Connect (System.Net.EndPoint remoteEP) [0x00000] in <filename unknown>:0 
                        at System.Net.Sockets.TcpClient.Connect (System.Net.IPEndPoint remote_end_point) [0x00000] in <filename unknown>:0 
                        at System.Net.Sockets.TcpClient.Connect (System.Net.IPAddress[] ipAddresses, Int32 port) [0x00000] in <filename unknown>:0 `

This error happens every time I try and establish a connection. Here's the code on the server side (drive.py file):

`import base64 #for lossless encoding transfer
from datetime import datetime #to set frame timestamp
import os #write + read files
import numpy as np
import shutil
import socketio #server
from flask import Flask #framework for web devices
from io import BytesIO #manipulate string and byte data in memory
import eventlet
import eventlet.wsgi 
import cv2

import tensorflow as tf
import keras
from keras.models import load_model
from PIL import Image

height = 320
width = 160     

def resize(image):
    return cv2.resize(image, (width, height), cv2.INTER_AREA)

#server init
sio = socketio.Server(always_connect = True )
#flask web app
application = Flask(__name__)

#init empty model and image array
net = None
image_array_before = None

#Speed limits
max_speed = 30
min_speed = 10

speed_limit = max_speed

#Server event handler
@sio.on('telemetry')
def telemetry(sid, data):

    if data:
        steering_angle = float(data["steering_angle"])
        throttle = float(data["throttle"])
        speed = float(data["speed"])    
        image = Image.open(BytesIO(base64.b64decode(data["image"])))
        
        #save frame
        timestamp = datetime.utcnow().strftime('%Y_%m_%d_%H_%M_%S_%f')[:-3]
        image_filename = os.path.join(r'path', timestamp)
        image.save('{}.jpg'.format(image_filename))
        
        try:
            image = np.asarray(image)
            image = resize(image)
            image = np.array([image])

            steering_angle = float(net.predict(image))

            global speed_limit
            if speed > speed_limit:   
                speed_limit = min_speed
            else:
                speed_limit = max_speed
            throttle = (1.0 - steering_angle**2 - (speed/speed_limit)**2)

            print ('{} {} {}'.format(steering_angle, throttle, speed))
            send_control(steering_angle, throttle)

        except Exception as e:
            print (e)

    else:
        
        sio.emit('manual', data={}, skip_sid = True)

@sio.on('connect')
def connect(sid, environ):
    print("connect ", sid)
    send_control(0,0) 

def send_control(steering_angle, throttle):
    sio.emit(
        "steer",
        data = {
            "steering_angle": steering_angle.__str__(),
            "throttle": throttle.__str__()
        },
        skip_sid = True)

if __name__ == "__main__":
    net = load_model('path')
    application = socketio.Middleware(sio, application)
    #deploy
    eventlet.wsgi.server(eventlet.listen(('localhost', 4567)), application)`

I've tried to fix it via disabling the firewall and changing the hostname, but to no avail. Any idea what might be wrong? Thanks!

techwithanirudh, 2021-01-19:
See this url about cv2 and ml https://youtu.be/mVUrErF5xq8

techwithanirudh, 2021-01-19:
and if it doesn't work try install version 2

techwithanirudh, 2021-01-19:
of the Simulator

haisenberg996, 2021-02-04:
> 并尝试降级到python-socketio 4.2.1
感谢感谢，成功解决了我的问题

haisenberg996, 2021-02-04:
> 并尝试降级到python-socketio 4.2.1

很好奇，您是怎么想到这个解决方法的呢？

thetechdude124, 2021-02-04:
Thank you all so much! Downgrading socket io worked!

Subhainr, 2021-02-05:
> When running the drive.py and simulator files, the connection is never established - it simply says "accepted", rather than actually connecting. When I looked at the output log for the simulator, [...]

pip install python-engineio==3.13.2
pip install python-socketio==4.6.1
 
let me know. do in your created environment only

sumit-mandal, 2021-02-12:
> pip install python-engineio==3.13.2
> pip install python-socketio==4.6.1
>
> let me know. do in your created environment only

your method worked. I was stuck for almost 2 days. You are saviour brother

mikel-brostrom, 2021-03-15:
Downgrading the libraries as explained [here](https://github.com/udacity/self-driving-car-sim/issues/131#issuecomment-774005040 ) worked for me! Thank you so much!

Subhainr, 2021-03-17:
> your method worked. I was stuck for almost 2 days. You are saviour brother

Thank u vaiya, for ur kind words. Move on

Subhainr, 2021-03-17:
> 

U r Welcome!!

Subhainr, 2021-03-17:
> your method worked. I was stuck for almost 2 days. You are saviour brother

> > 
> 
> U r Welcome!!

> Downgrading the libraries as explained [here](https://github.com/udacity/self-driving-car-sim/issues/131#issuecomment-774005040) worked for me! Thank you so much!

U r welcome!!

Subhainr, 2021-03-17:
> Downgrading the libraries as explained [here](https://github.com/udacity/self-driving-car-sim/issues/131#issuecomment-774005040) worked for me! Thank you so much!

U r Welcome!!

AserElkhateeb99, 2021-05-31:
> and try downgrading to python-socketio 4.2.1
how you did it?

hpt1988, 2021-07-11:
`pip3 install “python-socketio<4.3” “python-engineio<3.9` is the solution

githubrandomuser2017, 2021-07-25:
> When I looked at the output log for the simulator,

@thetechdude124 Where did you find the simulator's output log? I'm running the simulator on a Mac.

HaoranZhuExplorer, 2021-08-02:
solved my problem. thanks!

tensorway, 2021-08-03:
solved my problem. thanks!